sinara-hw/meta #77: Simple 10W RF amplifier

gkasprow, 2023-03-22:
We need to build a 10MHz amplifier delivering 10W of power to drive the RF resonator, which will boost it to 500V or so.
The amplifier will be 6HP wide and work with EEM and DI/OT.
The initial specification:
- single channel (unlike SiLPA)
- 6 (DI/OT) compatible or 8 HP (EEM) wide
- Push-pull configuration using 2 MOSFETs (very popular within the SSB community)
- 10-100MHz or wider 
- 10W output power in linear mode (let's say 2nd and 3rd harmonic <30dBc)
- Drive stage using either fast opamp or mini circuits gain block
- Must work with Urukul, gain ~40dB
- High linearity; the design will let the user choose high efficiency or high linearity mode by adjusting the bias current
- RF power detector for feedback (RF amplitude stabilisation)
- Basic diagnostics - current measurement for both output stage MOSFETs (symmetry is critical for linearity); output power measurement
- Overtemp protection (LM75)
- Fast output interlock, no directional coupler foreseen

Would you have any thoughts to share about it?

pathfinder49, 2023-03-23:
What is the intended use-case for such an amplifier? If this is supposed to be a "convenient amplifier for general use" then I think requirements are a bit different to applications that require high amplitude stability of the RF chain.

gkasprow, 2023-03-23:
In our application of generating RF field for anti-matter experiments, it's sufficient.
What would you change to make it suitable for your use case?

dtcallcock, 2023-03-23:
If you want to do something really fast and simple, you could use 8x [ZX60-100VH+](https://www.minicircuits.com/WebStore/dashboard.html?model=ZX60-100VH%2B) in parallel.  I've been very happy using this for trap drive.  We only need 1W for our trap right now, but anticipating needing more in future we built a 4x version to get to 4W - worked well on the bench.

dhslichter, 2023-03-23:
@dtcallcock using the quadrature hybrid "add amplifier outputs" trick?

gkasprow, 2023-03-24:
The issue with these amp bricks is also their availability. I'd need 10 of them at least to get 40dBm at 1dB Compression.
I can, of course, DYI sth, but I'd prefer to put some effort and make module which works for other use cases. I'm not in a hurry.

pathfinder49, 2023-03-24:
> In our application of generating RF field for anti-matter experiments, it's sufficient. What would you change to make it suitable for your use case?

In my usecase we cared about fractional gain stability at <10^-5. Which means stable voltage references and the ability to heating everything well in a temeprature stabilised box. This porbably would make things needlessly complicated for a general purpose design (as this seems to be).

gkasprow, 2023-03-25:
I asked physicists, and they said the higher the stability, the better. Let's see what we can achieve without complicating things much.

gkasprow, 2023-03-25:
@pathfinder49 Are you using some active RF amplitude stabilisation system?

pathfinder49, 2023-03-25:
The setup I used has a passive fractional amplitude stability $<2\times10^{-5}$. Careful temperature stabilisation and reducing cabling length were quite important for this. (Though the temperature stability may have been over engineered in my setup.) I don't think a rackmount EEM card would be a good fit for this kind of application. In case some of it proves useful, I'll detail the setup below.
1. Pre-amplification, the amplitude stable signal was [generated from stable DC voltage rails and a logic-level converter](https://github.com/OxfordIonTrapGroup/Squareatron-5000). The board was heatsunk to a metal enclosure which was temperature stabilised within $0.001\degree \mathrm{C}$ (using [Thermostat](https://github.com/sinara-hw/Thermostat)). This box was kept small to avoid convection air-currents within the enclosure. The box was wrapped in insulating foam and placed in another metal box. The important bit here is to keep a stable temperature distributiopn through the board. As the RF signal is steady-state, we acheive the stable temperature distribution by temperature stabilising the environement of the board.
2. The stable RF signal was amplified using an API, BXMP1042 (powered by Wenzel, LNVR-24-15-5-1). Both the amplifier and its supply were temperature stabilised in the same manner as described in 1.

dhslichter, 2023-04-04:
Agreed that most ion trap QIP experiments would not want the trap rf sourced from an EEM card.

dtcallcock, 2023-04-04:
> I'd need 10 of them at least to get 40dBm at 1dB Compression.

Yeah, TBF this makes a lot more sense at 1-4W.  10W is just not a great place to be for your trap drive but I guess they have some giant trap.

gkasprow, 2023-04-04:
yep, it's quite big
[image]