sinara-hw/meta #80: Opening a design file from the community

constancekoh1, 2023-06-24:
Hello, I am new to Circuit Maker. I have been trying to open the circuits in this link (https://workspace.circuitmaker.com/Projects/Details/n-yj/AUV-4-0-SIGNAL-CONDITIONING) but have failed to do so. I have already installed the Circuit Maker application as well as signed up for an Altium account. Does anyone know how I should proceed?

constancekoh1, 2023-06-24:
If I try opening the file by clicking on "Open in Altinum 365" it redirects me to a page that says " This page isn’t working365.altium.com redirected you too many times.
[Try clearing your cookies](https://support.google.com/chrome?p=rl_error&hl=en-US).
ERR_TOO_MANY_REDIRECTS"

jordens, 2023-06-24:
I don't see how this has anything to do with sinara.
Maybe check with circuit maker and/or the project you are interested in.

closed 2023-06-24 by jordens