sinara-hw/meta #81: RFC: very low latency PID

gkasprow, 2023-10-27:
A few times, I was asked about the feasibility of building something similar to a very low latency PID like [FALC-PRO](https://www.toptica.com/products/laser-rack-systems/laser-locking-electronics/falc-pro)
We are talking about a delay of <10ns at >50MHz BW.
We can start working on it, providing sufficient interest in the community.

dtcallcock, 2023-10-27:
Is there a way of adding these features to: https://github.com/sinara-hw/Fast_Servo/wiki ? Given firmware support will probably the bottleneck, 1 board is more likely to be supported than 2.

gkasprow, 2023-10-28:
Exactly. There are no converters on the market able to reach 10ns round trip delay. So this must be analog with digital tuning only

dtcallcock, 2023-10-29:
Could it be a mezzanine board then?

gkasprow, 2023-10-29:
> Could it be a mezzanine board then?

sure. Even for Stabilizer. I see it as stand-alone device.

gkasprow, 2023-11-27:
I found very interesting[ paper](https://www.researchgate.net/publication/318662587_CFOA-based_fractional_order_PIlDd_controller) on high-speed PIDs using current feedback amplifiers 
P, I, and D gain/slope can be tuned by voltage using OPA860 transconductance.
[image]

gkasprow, 2023-11-27:
Yet [another](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC4897109/) approach to tunable integrator

gkasprow, 2023-11-27:
@sbourdeauducq, what would be the primary use case? Make it a faster version of Pounder, replace digital PID with analogue one? Or make it a general-purpose, very low-latency regulator, optionally with an RF mixer at the input?
The main issue is who will test our prototypes in a real live scenario.

gkasprow, 2023-11-28:
We can always make PI^2D^2

gkasprow, 2023-11-28:
Or even PI³D² . The challenge would be tuning such regulator.

gkasprow, 2023-12-09:
I played a bit with integrators and differentiators based on OTA concept.
It looks interesting.
First approach:
[image]

[image]

the second approach is interesting because the differentiator can have grounded capacitor
[image]
[image]

Initially, I thought that OTA enables tuning the integrator/differentiator slope by changing the Iq.
But it works only when we use diamond transistor Ic/Ub transconductance. It means we have to operate in the +/-20mV linear range of the bipolar transistor base.
So that approach will work only for such integrators
[image]

However, the non-linearities of the input stage cause distortions.
Minimal transconductance - output voltage is 1Vpp
[image]
Maximal transconductance - output voltage is 7.5Vpp
[image]
This example shows that, in principle, the bias current can change the gain of the integrator.
I am still trying to figure out how to make the differentiator in such a way that tunable transconductance would work.

gkasprow, 2023-12-09:
Of course, one can use VGA in front of a classical integrator/differentiator, but I'm not sure if that's what we are looking for. That does not change the slope but only the gain.

gkasprow, 2023-12-09:
In the paper I pointed before there is an example of tunable PID that works according to this principle
[image]
The CFOA consist of two OTA
[image]
The principle is identical to my simulations. The authors say that the transimpedance can be controlled by Iq.
That's interesting because a diamond transistor should always have the Ic/Ie (common base config) current gain =1.
I've just noticed that are using the common emiter approach. And indeed, we can tune the slope! The range is much narrower than in the case of voltage steering. The circuit:
[image]

Maximum Iq, Vout = 443mVpp
[image]
Minimum Iq, Vout = 728mVpp. We operate in a linear region because the transistor works in the current mode.
[image]

So, now, let's swap the capacitor with the resistor and see how the integrator behaves.

[image]
Maximum Iq, Vout = 486mVpp
[image]

Minimum Iq, Vout = 480mVpp, nearly no difference
[image]

gkasprow, 2023-12-09:
To conclude:
- it's possible to build a tunable integrator (voltage steering needed, distortion,  1:7 range tuning)
- it's possible to build tunable differentiator (current steering, nearly 1:2 range tuning)
- let's compare this approach to the classical one in terms of noise.

gkasprow, 2023-12-09:
Interesting finding.
I expected the current gain to be close to 1, like in the common-base transistor.
But in the case of diamond transistors, that's not true.
Let's take such a schematic.
[image]
Min Iq, Gain = 0.84
[image]
Max Iq, Gain = 0.55
[image]

We have the same resistor at the input and the output, so we have some currents (E is virtual ground).
Moreover, the higher the gain we get, the lower the transconductance.

gkasprow, 2023-12-09:
I replaced OPA660 with OPA860. Tham main difference is that the latter has Iq adjusted only for OTA, not for the output buffer.
And when I chance the Iq, the gain stays constant.
The schematic
[image]

gkasprow, 2023-12-09:
I built the differentiator with OPA860, but the slope doesn't change with the Iq, unfortunately. It's, however, conforming to the theory of OTA.
[image]

gkasprow, 2023-12-09:
Since OPA660 is NRE, we cannot really make the differentiator with OPA860 because OPA660's bugs were fixed.

gkasprow, 2023-12-11:
The only option is to use the approach of commercial PID controllers: variable resistors + switched capacitor banks.
FALC Pro is also using relays and RC banks. It's visible in the configuration software that all PID parameters are selectable, for example:
[image]

gkasprow, 2023-12-11:
RC switching can be done using muxes (DG series from Analog), lateral DMOS switches (SST21x) or opto-MOS.
We used OPTO-MOS in high-speed analogue front ends for CERN high-performance ADC cards. Here is an example of how to build a 100MHz variable range amplifier. The design is on the OHWR webpage.
[image]

I want to build sth better than FALC Pro. They use just a few discrete values, so why not use 2^n weights? This gives far more flexibility. Moreover, I'd consider adding a varicap to tune the cutoff frequency continuously. That could be important when compensating very high speed loops. 
Let's assume we have integrator capacitor values:
1. 100p,
2. 200p
3. 400p
4. 800p
5. 1.6n
6. 3.2n
7. 6.4n
8. 12.8n
9. 25.6n
10. 51.2n

The number of switches is similar, but the tuning range is continuous from 100p to 100nF.
When we add a varicap that can tune from 10p to 200pF but only use a linear range of 10..100pF, we end up with a range from 10pF to 100nF with 1pF increment! That's four orders of magnitude. Of course, the monotonicity of such a capacitor would be an issue, but when needed, it can be fixed in software.

gkasprow, 2023-12-12:
Let's discuss the tuning ranges. As a first approach, I'd use ones from Toptica documentation. It's available online as a zip package of firmware for DLC Pro.
I1: 1 = 1.5 kHz; 2 = 3.0 kHz; 3 = 7.0 kHz; 4 = 15 kHz; 5 = 31 kHz; 6 = 70 kHz; 7 = 130 kHz; 8 = 290 kHz; 9 = 620 kHz; 10 = 7.0 MHz
I2: 1 = 25 Hz; 2 = 50 Hz; 3 = 100 Hz; 4 = 220 Hz; 5 = 470 Hz; 6 = 1.0 kHz; 7 = 2.2 kHz; 8 = 5.0 kHz; 9 = 10 kHz
I3: 1 = 0.6 Hz; 2 = 1.8 Hz; 3 = 6.0 Hz; 4 = 18 Hz; 5 = 60 Hz; 6 = 180 Hz; 7 = 600 Hz
I1: 1 = 10 kHz; 2 = 20 kHz; 3 = 40 kHz; 4 = 90 kHz; 5 = 190 kHz; 6 = 400 kHz; 7 = 760 kHz; 8 = 1.5 MHz; 9 = 3.5 MHz; 10 = 7.2MHz
I2: 1 = 10 kHz; 2 = 20 kHz; 3 = 45 kHz; 4 = 100 kHz; 5 = 200 kHz; 6 = 420kHz; 7 = 700kHz; 8 = 1.2MHz; 9 = 3.5MHz; 10 = 6.0MHz

Moreover, they have input gain selectable 1x or 5x
Then, we have the detector offset
And one gain parameter at the output of the regulator.

gkasprow, 2023-12-12:
Do you have any idea what such PI^2D^3 regulator architecture should look like?
We have two differentiators, three integrators and one proportional term.
 We can arrange them in several ways.
For example, classical PID, then I, then ID or reverse order. What is used in the labs to compensate for very high bandwidth loops?

hartytp, 2023-12-15:
Not off the top of my head, no. Sorry.

jordens, 2023-12-18:
I haven't worked on these analog implementations beyond textbook PID.

gkasprow, 2023-12-18:
An example transfer  function could help

jordens, 2023-12-18:
Again, I haven't used anything analog beyond PID, so I can only speculate: about one to two decades per behavior, as in the values you quote

* I² from 0 to 10 Hz
* I to 1 kHz
* P to 100 kHz
* D to 1 MHz
* turning into some intrinsic low-pass above that

One can speculate whether an analog design can actually achieve high-fidelity I³ behavior at relevant frequencies in practice.

Other transfer functions like notches, gain limits (I and D), and MIMO controllers might be much more interesting in practice than high order I/D.

In the absence of proper requirements, specifications, etc this is all wild speculation.

gkasprow, 2023-12-18:
I found in a few papers that use FALC pro that the integrators are not unlimited. There is one unlimited integrator but as an additional circuit. Essentially FALC Pro is a tunable filter where you can switch two hipass and three lowpass circuits. It's not real PI3D2 controller. The first stage is limited PID but two other stages are just lowpass and highpass correction filters which act on top of the original characteristics and can be switched off. I have this box as a part of coherent 729 laser in my ion trap setup.