sinara-hw/meta #88: CoaxPress support

gkasprow, 2024-03-20:
We get more and more requests for support of CoaxPress cameras.
There are several options:
1. Use EEM FMC carrier and CoaxPress FMC from Kaya
2. Make a dedicated DIOT card with simple CoaxPress Rx and stream data using MGT lane (requires Kasli-DIOT, which have MGT routed to DIOT slots)
3. Make a simple CoaxPress to SFP adapter. The only issue is that CoaxPress Rx requires an uplink LVCMOS line. So, it has to be connected to FPGA IO. We will do it in Kasli-DIOT (we will use the RATE_SEL line), but that approach won't work for the existing Kasli. 
4. We could use CoaxPress to SFP adapter and MGT Tx line to stream 40Mb/s uplink data but that would require some extra work on firmware but would keep compatibility with all Kasli versions

nkrackow, 2024-05-07:
We were also evaluating CoaXPress cameras but in addition to the closed standard (not sure how important with an external IC) the camera manufacturers have proprietary settings/telemetry channels via CoaXPress that you'd have to reverse engineer for any camera since they usually want to sell their custom flavored frame grabber as well... 
Also electrically the interface looks more annoying than shielded twisted pairs in one cable like you see in all other high-speed digital interfaces.

gkasprow, 2024-05-14:
@sbourdeauducq We have ~20 engineers who use Altium. Three of them, including me, know Kicad. We are far more productive with Altium than with Kicad.
It makes no economic sense to shift to a much worse tool. That would require training - one gets productive after two months with Kicad (we use it for clients that pay for it, i.e. CERN). Just multiply 17 *2 * engineer salary. And then add +100% of time spent on every project.
Another issue is SI/PI verification, which doesn't work well for Kicad. We use Hyperlynx/SIWave for nearly every project.
Altium is cheap. It is cheaper than one month for an engineer to work half-time.
Moreover, we get massive discounts on Ansys and Altium tools at the WUT.
Finally, we use excellent Altium CERN libraries. After translating it to Kicad, they need a lot of manual work.
So unless someone pays us extra to design in Kicad, we will continue with Altium.

gkasprow, 2024-05-14:
your comment makes no sense at all.

gkasprow, 2024-05-14:
Why does the Altium license prohibit us from having nice things? We are implementing all the changes and fixes the community requires. We happily produce nearly every piece the HW community asks for—for free—because we have public funding and our own time.
Viewing does not require a license - you can do it even online.
You can post your designs using whatever tool you want.
You can also make Altium - Kicad translation and continue with Kicad.
I often take Kicad designs and convert to Altium because it's far more productive for me.
So just because you want, we should make a massive effort and lower our productivity...
Would you switch your entire toolchain to another just because I want to? 
We have to live with the tools we have. Perfect is the enemy of good

gkasprow, 2024-05-14:
Of course this latchup will be fixed in next revision.

gkasprow, 2024-05-14:
I can give you temporary access to the license if that helps.