sinara-hw/meta #89: Zotino & Fastino FP connector in DIOT

marmeladapk, 2024-03-27:
We're thinking about changing front panel connector on Fastino and Zotino in DIOT versions to mini SAS (the same ones that are in Shuttler, maybe we could even use Shuttler frontend). They have much lower crosstalk that has been problematic in case of Fastino and even long cables aren't expensive. That way we could avoid adapters hanging on the FP. This would mean that accompanying adapters (HD68 and possibly SMA/BNC) would also be changed for DIOT.

kaolpr, 2024-03-29:
How do you want to place adapters / mezzanines (e.g. HVAMP32)? External non-DIOT crate?

gkasprow, 2024-03-29:
HVAMP32 fits nicely within 6HP DIOT Panel
[image]

gkasprow, 2024-03-29:
In the case of mini SAS plugs I designed an adapter that outputs 8 u.fl to SMA pigtails
[image]

gkasprow, 2024-03-29:
They nearly fit. One would have to shift the HVAMP connectors 1mm to the back of the board
[image]

kaolpr, 2024-03-29:
Perfect! I don't think there are many HVAMPs out there right now and anyway they'll go combined.

However, I think that some more user-resistant adapter (e.g. a box connected with cable, like Shuttler AFE but with SMA) would be nice. It could be used separately from the pig-tail adapter which could be nice for some quick testing.

gkasprow, 2024-03-29:
Yep. We can make the same kind of adapter that was made for HD68. I mean one with BNCs and MCX.

gkasprow, 2024-03-29:
When we replace 3M four wall headers with standard gold-pins, all will fit
[image]

kaolpr, 2024-03-29:
It looks like this solution will somehow compromise ribbon usage as with goldpins it will be possible to connect it the wrong way. Maybe it's better to evaluate how many HVAMPs are there and see if we really need to care?

gkasprow, 2024-04-04:
We make new Fastino release to fit new, longer HVAMP :)

gkasprow, 2024-04-04:
We can also shift the connectors and loose compatibility with the first batch of HVAMPs

kaolpr, 2024-04-04:
I believe this is not a big problem.

gkasprow, 2024-04-05:
pls make sure we have same pinout of SPI interface in Shuttler