sinara-hw/meta #94: low noise DC/DC converters for fast IO boards

gkasprow, 2024-12-24:
We got several questions about noisy DIO modules.
The issue is the capacitance between the switching transformer winding and floating ground. For TMH1205S it can be as high as 100pF
That can be mitigated by increasing the capacitance between floating GND and chassis, but that would also lower the isolation barrier impedance.
I plan to implement improved LLC converters with 0.5pF coupling capacitance. That adds complexity and cost, and probably not everybody needs them, so the idea is to make them footprint-compatible with TMH1205S, and one could order standard or low-noise DIO modules

hartytp, 2024-12-24:
> That adds complexity and cost, and probably not everybody needs them, so the idea is to make them footprint-compatible with TMH1205S, and one could order standard or low-noise DIO modules

What is the actual cost and complexity level we're talking about here?

Can I offer an alternative view on this? My guess (subject to an answer to the above) is that, at the volumes people are working at, the cost of support is greater than the BoM cost. Also, most users probably care more about having something which "just works" with good levels of performance, than they do about small optimizations to the BoM cost (the cost of debugging a noise issue in the lab is generally far greater than the cost of low noise parts).

So I'd suggest that either we think this is worth doing, in which case, once we've thoroughly prototyped it, we should make this the default (if not the only) supported option. If we think this is a niche thing then we shouldn't clutter up our designs with it. If we keep stacking up variants, the layouts are going to become a mess and the challenge of supporting users grows exponentially ("I can't help you debug this because I don't have the particular combination of hw variants you're using to hand").

gkasprow, 2024-12-24:
I think you are right.
We will add a 7 EUR transformer, a 2 EUR controller, an optocoupler, and MOSFETs, making an extra ~20 EUR per converter.
So, make the default option or create a new product, like DIO-SMA-Low-Noise. In the case of a 16-channel DIOT module, that would add 80 EUR to the BOM.

gkasprow, 2024-12-24:
Of course, TMH1205S is 7EUR/pc so the difference wouldn't be that high. Let's test it and make a default option, at least for one type of IO module.

gkasprow, 2025-02-18:
I designed low noise DC/DC and sent for manufacturing

[image]

gkasprow, 2025-02-19:
@jordens sure

[Ultra_LN_ISOL_SMPS.zip](https://github.com/user-attachments/files/18864050/Ultra_LN_ISOL_SMPS.zip)

gkasprow, 2025-02-19:
It's modified TI design + feedback which the IC designer did not foresee :)

gkasprow, 2025-02-19:
I already noticed some bugs, so wait until I make it run.

MorganTL, 2026-09-09:
> I designed low noise DC/DC and sent for manufacturing

Btw what's the reasoning behind adding a common mode choke on the primary side?

MorganTL, 2026-09-29:
The NXE2S1205MC (the alternative DCDC converter on SMA-DIO) only has 2.1 pF isolation capacitance and offer much better noise performance when Fastino / Zotino ground is connected to the SMA-DIO ground from my latest measurement (https://git.m-labs.hk/morgan/ln_iso_smps/issues/1#issuecomment-63161)

If we want to an alternative converter with single digit isolation capacitance and 2W output power, we can use IHL02 series from XP power with a typical isolation capacitance of 7 pF. (Not footprint compatible to TMH1205S)